stepjam/PyRep #280: RuntimeError: Handle Panda_gripper_attachProxSensor does not exist.

deepdad, 2021-06-19:
I get the error:

```
RuntimeError: Handle Panda_gripper_attachProxSensor does not exist.
```
when running:
```from pyrep import PyRep
from pyrep.robots.arms.panda import Panda
from pyrep.robots.end_effectors.panda_gripper import PandaGripper
import time

pr = PyRep()
# Launch the application with a scene file in headless mode
# pr.launch('PyRep/tests/assets/test_scene.ttt', headless=False)  # has an error
pr.launch('PyRep/examples/scene_panda_reach_target.ttt') #scene_with_panda.ttt')
pr.start()  # Start the simulation

# normally at the top
from pyrep.objects.shape import Shape
from pyrep.const import PrimitiveShape
# change the scene
r = g = b = 1
w = h = d = 100
x = y = z = 50

print("no sure this works, I needed to move the GUI bar above the logged output at the bottom to see the 3D scene")
object = Shape.create(type=PrimitiveShape.CYLINDER,
                      color=[r,g,b], size=[w, h, d],
                      position=[x, y, z])
object.set_color([r, g, b])
object.set_position([x, y, z])

time.sleep(5)

print("no sure this works")

# use the Panda robot arm
arm = Panda()  # Get the panda from the scene
gripper = PandaGripper()  # Get the panda gripper from the scene

velocities = [.1, .2, .3, .4, .5, .6, .7]
arm.set_joint_target_velocities(velocities)
pr.step()  # Step physics simulation

done = False
# Open the gripper halfway at a velocity of 0.04.
while not done:
    done = gripper.actuate(0.5, velocity=0.04)
    pr.step()

pr.stop()  # Stop the simulation
print("wait a second to copy the log")
time.sleep(10)
pr.shutdown()  # Close the application
```

Elissa0723, 2021-07-13:
I get the same error, did you fix it?

deepdad, 2021-07-14:
No. I use:
```
"""
A Franka Panda reaches for 10 randomly places targets.
This script contains examples of:
    - Linear (IK) paths.
    - Scene manipulation (creating an object and moving it).
"""
from os.path import dirname, join, abspath
from pyrep import PyRep
from pyrep.robots.arms.panda import Panda
from pyrep.objects.shape import Shape
from pyrep.const import PrimitiveShape
from pyrep.errors import ConfigurationPathError
import numpy as np
import math

LOOPS = 10
SCENE_FILE = 'PyRep/examples/scene_panda_reach_target.ttt'
#' join(dirname(abspath(__file__)), 'scene_panda_reach_target.ttt')
pr = PyRep()
pr.launch(SCENE_FILE, headless=False)
pr.start()
agent = Panda()

# We could have made this target in the scene, but lets create one dynamically
target = Shape.create(type=PrimitiveShape.SPHERE,
                      size=[0.05, 0.05, 0.05],
                      color=[1.0, 0.1, 0.1],
                      static=True, respondable=False)

position_min, position_max = [0.8, -0.2, 1.0], [1.0, 0.2, 1.4]

starting_joint_positions = agent.get_joint_positions()

for i in range(LOOPS):

    # Reset the arm at the start of each 'episode'
    agent.set_joint_positions(starting_joint_positions)

    # Get a random position within a cuboid and set the target position
    pos = list(np.random.uniform(position_min, position_max))
    target.set_position(pos)

    # Get a path to the target (rotate so z points down)
    try:
        path = agent.get_path(
            position=pos, euler=[0, math.radians(180), 0])
    except ConfigurationPathError as e:
        print('Could not find path')
        continue

    # Step the simulation and advance the agent along the path
    done = False
    while not done:
        done = path.step()
        pr.step()

    print('Reached target %d!' % i)

pr.stop()
pr.shutdown()
```

petrvancjr, 2021-09-07:
Hi @deepdad, 
open **scene_panda_reach_target.ttt** scene. Change the object name `Panda_gripper_grasp_sensor` to `Panda_gripper_attachProxSensor` by double tapping the name on _Scene hierarchy_ tree and save the scene.
You will probably also need to change object the name `Panda_gripper_attach_point` to `Panda_gripper_attachPoint`.

closed 2021-11-14 by stale[bot]