stepjam/PyRep #285: [Urgent] CoppeliaSim 4.1 expired

MRzNone, 2021-07-16:
When I opened CoppeliaSim, it said V4.1 is expired.  Is there a plan to officially support 4.2?

Related to https://github.com/stepjam/PyRep/issues/284

[image]

frietz58, 2021-07-16:
A hacky, non-ideal way that bypasses this is to change the system time, e.g. one year back. At least on Ubuntu 18.04, I could start CoppliaSim 4.1, either directly or through PyRep, that way:
[image]

MRzNone, 2021-07-16:
Thanks for the response.  I could use pyrep by installing it with v4.2 (or hacks like yours).  I am just wondering if there will be official support.

stepjam, 2021-07-16:
Thanks for bringing this to my attention. I'll get to work on doing the update today.

MRzNone, 2021-07-16:
Thank you very much!

deepdad, 2021-07-17:
I don't see any problems really with TargetReach. Except this, but that already existed in 4.1:

[image]

deepdad, 2021-07-17:
Interesting that Coppelia didn't notify you earlier. Maybe you should get them to ping you when this kind of thing is set to happen.

Coppelia, 2021-07-19:
Hello,

at Coppelia Robotics, we are continually introducing new features and better, improved ways of doing things in CoppeliaSim. For this reason, it is important to continually update/upgrade to the newest CoppeliaSim version. We thought that the best way of doing this would be to limit the runtime of Edu versions for exactly one year, but it seems that many users feel frustrated to be forced to move to the newest CoppeliaSim Edu release.

You have two alternatives:

1. If you want to Continue to use CoppeliaSim V4.1.0, then simply add following in the system/usrset.txt file:

allowOldEduRelease=7501

2. Alternatively, you can also upgrade to CoppeliaSim V4.2.0, and edit the value of showOldDlgs (in the same file) to true: this will enable old, deprecated dialogs. As of now, it appears that there is also a problem with simGetConfigForTipPose in CoppeliaSim V4.2.0, so PyRep users better use the first alternative for now.

Cheers

stepjam, 2021-07-19:
**To PyRep users:**
This is now resolved. Simply rerun `pip install .` (after pulling from master)
**There is no need to redownload CoppeliaSim 4.1**

closed 2021-07-19 by stepjam