stepjam/PyRep #294: QObject::~QObject: Timers cannot be stopped from another thread

NikoFeith, 2021-10-19:
Ubuntu 20.04
CoppeliaSim 4.2 EDU
Python 3.8

I tried to run the example "example_panda_end_effector_control.py" after 1 or 2 seconds CoppeliaSim stops and closes.
Afterwards the terminal returns:
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread

ryanreadbooks, 2021-11-02:
I encountered the same question when I tried to run the examples under Ubuntu 16.04 + CoppeliaSim 4.2 EDU + Conda 4.10.3 + Python 3.7.

when running "example_panda_reach_target.py", output is below
```Reached target 0!
Reached target 1!
Reached target 2!
Reached target 3!
Reached target 4!
Reached target 5!
Reached target 6!
Reached target 7!
Reached target 8!
Reached target 9!
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread

closed 2022-01-09 by stale[bot]

Leilasjd, 2022-09-17:
Hi, can you fix this problem? I get the same error. If you know what I should do plz let me know. Tnx