stepjam/PyRep #302: test_get_nonlinear_path() failed for Panda

ZhekaHauska, 2022-01-17:
**Coppeliasim** version: **4.1.0** for Ubuntu 20.04
**PyRep** version: **4.1.0.2** cloned from github
**Python** version: **3.9**

I've run [tests/test_arms_and_configuration_paths.py](tests/test_arms_and_configuration_paths.py) and test_get_nonlinear_path() has failed with error:

```
[sandboxScript:info]   Simulator launched, welcome!
[sandboxScript:info]   Simulation started.
[PyRep@addOnScript:error]   251: IK group does not exist. (in function 'sim.getConfigForTipPose')
    stack traceback:
        [C]: in function 'getConfigForTipPose'
        [string "PyRep@addOnScript"]:251: in function '_findSeveralCollisionFreeConfigsAndCheckApproach'
        [string "PyRep@addOnScript"]:357: in function <[string "PyRep@addOnScript"]:338>
[CoppeliaSim:error]   External call to simCallScriptFunction failed (getNonlinearPath@PyRep): Error in script function.

```

closed 2022-03-27 by stale[bot]

SamNPowers, 2023-04-05:
I had this issue too. For me, at least, the issue was that I had installed PyRep when my COPPELASIM_ROOT was pointing at the wrong version of Coppelia (4.5). When I switched to the right version (4.1), I failed to also reinstall PyRep, and this was the error I encountered. Re-running `pip install -e .` in PyRep after making sure the ROOT was properly set fixed it for me.

zwbx, 2024-04-05:
This error does not happens all the time, When I use EndEffectorPoseViaPlanning and a manipulation policy network that is trained by myself, the error sometimes shows. When I use action given by the RLBench demo, there is no error. That means the present of the error seem to be related to the action. I make sure that the action format is correct because in some case everything goes well.