stepjam/PyRep #304: In what frame of reference is this Jacobian defined?

sageshoyu, 2022-01-25:
https://github.com/stepjam/PyRep/blob/8bf26f47e6abc098bf4961f20cfe122d54e611e6/pyrep/robots/arms/arm.py#L459

I'm currently implementing a variable-impedance operation-space controller and am using this Jacobian to convert end-effector wrenches to joint torques. In case I need to perform any frame changes for my computations - what is this Jacobian's frame of reference? End-effector body? Global frame?

Thanks!

Coppelia, 2022-01-31:
@sageshoyu: the end-effector frame can be defined relative to any reference frame, this depends on the settings of that IK group in CoppeliaSim. But most of the time this is relative to the world frame

sageshoyu, 2022-02-02:
Thank you so much. Is there a programmatic way of acquiring an IK group's frame of reference via the PyRep API?

Coppelia, 2022-02-02:
I think there is no API in PyRep API (we are not the creator/maintainer of PyRep, so we are not that familiar with it). As far as I understand PyRep still relies on CoppeliaSim V4.1, while CoppeliaSim V4.3 is now available. In CoppeliaSim V4.2+, IK task creation is completely handled via code. In CoppeliaSim V4.1 and prior to that, an IK task is created via the GUI. So you will have to open the scene and inspect the IK task via the GUI [Menu bar --> Tools --> Calculation module properties --> Kinematics]. Select the IK group/task, then click "Edit IK elements". Select an IK element and you can see "Relative to coordinate frame" which indicates relative to which coord. frame the constraints are computed.

closed 2022-02-08 by sageshoyu