stepjam/PyRep #313: CoppeliaSim v4.2 expiration

mch5048, 2022-04-16:
Hi, thanks for maintaining this amazing project! I really enjoy working on the pyrep and rlbench for by project.

While I using rlbench with CoppeliaSim 4.2, I encountered exactly the same license expiration issue that occured year ago with CoppeliaSIm4.1 #285 .

While I can move to CoppeliaSim 4.3 and newest version of rlbench, I don't have much time for migrating my project as of now😥.
As described in the issue #285, it seems parsing the `allowOldEduRelease` may resolve the solution, but I cannot guess the
corresponding version for CoppeiaSim4.2 (which was 7501 for CoppeliaSim4.1).

I wish I can get some help for this issue! maybe @Coppelia can give some advice?
Thanks in advance!

mch5048, 2022-04-16:
It seems my current code is compatible with CoppeliaSim4.3. To prevent future issues, hope some solution can be shared soon!

Coppelia, 2022-04-18:
@mch5048, have a look at the workaround described here: [https://forum.coppeliarobotics.com/viewtopic.php?f=9&t=9334](url)

mch5048, 2022-04-21:
@Coppelia Much appreciate for your help!

closed 2022-04-21 by mch5048