stepjam/PyRep #320: libcoppeliaSim.so.1: undefined symbol: _ZdlPvm, version Qt_5

JiahengZeng, 2022-07-06:
Hi, when I try to run the example code in the /example file, I meet this problem:
libcoppeliaSim.so.1: undefined symbol: _ZdlPvm, version Qt_5
Could you please help me? Cause I try to solve this for a long time, but have no idea.
Thank you very much.

JiahengZeng, 2022-09-14:
But when I try to Debug the code in Pycharm,  this error comes again.
I have no idea this time.

cpaxton, 2022-09-14:
Is Pycharm using your bashrc? You should be able to change the command you're using to debug code

JiahengZeng, 2022-09-22:
Actually, it does not work either RUN nor Debug. I changed to use Vs code, and solved. I have checked the env problem, but have no effective. I think there must be some problem that I haven't found.

closed 2023-01-05 by stale[bot]

15101051, 2025-03-12:
For anyone encountering this issue, you can check whether your conda environment contains any Qt-related packages (such as `qt-main`, `pyqt`, etc.). If so, try removing them. In my case, the issue was caused by OpenCV introducing Qt packages into the environment.

15101051, 2025-03-14:
> For anyone encountering this issue, you can check whether your conda environment contains any Qt-related packages (such as `qt-main`, `pyqt`, etc.). If so, try removing them. In my case, the issue was caused by OpenCV introducing Qt packages into the environment. I then use `pip install opencv-python` to install opencv.

I just found opencv-python will also introduce Qt libraries...So maybe `pip install opencv-python-headless` may help.

GuHuangAI, 2025-05-24:
> Actually, it does not work either RUN nor Debug. I changed to use Vs code, and solved. I have checked the env problem, but have no effective. I think there must be some problem that I haven't found.

You should add environmental variables in the debug configuration.