stepjam/PyRep #322: QObject::~QObject: Timers cannot be stopped from another thread

Vishal1711, 2022-08-05:
I tried to run  ```example_turtlebot_navigation.py``` from the example folder I faced the following error. How to solve this?

```
vishal@vishal-Prestige-14Evo-A11M:~/git_ws/AuE8230Spring22_VishalJadhav/catkin_ws/src/PyRep/examples$ python example_turtlebot_navigation.py 
Traceback (most recent call last):
  File "example_turtlebot_navigation.py", line 39, in <module>
    path = agent.get_nonlinear_path(position=pos, angle=0)
  File "/home/vishal/.local/lib/python3.8/site-packages/pyrep/robots/mobiles/nonholonomic_base.py", line 104, in get_nonlinear_path
    path = self._get_nonlinear_path_points(
  File "/home/vishal/.local/lib/python3.8/site-packages/pyrep/robots/mobiles/mobile_base.py", line 133, in _get_nonlinear_path_points
    _, ret_floats, _, _ = utils.script_call(
  File "/home/vishal/.local/lib/python3.8/site-packages/pyrep/backend/utils.py", line 65, in script_call
    return sim.simExtCallScriptFunction(
  File "/home/vishal/.local/lib/python3.8/site-packages/pyrep/backend/sim.py", line 698, in simExtCallScriptFunction
    _check_return(ret)
  File "/home/vishal/.local/lib/python3.8/site-packages/pyrep/backend/sim.py", line 27, in _check_return
    raise RuntimeError(
RuntimeError: The call failed on the V-REP side. Return value: -1
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex

```

Leilasjd, 2022-09-17:
I get the same error, what should I do?

closed 2023-01-05 by stale[bot]

zwbx, 2024-04-06:
same error, any updates?