stepjam/PyRep #327: Trying to run PyRep in a singularity container

Abetelgeusian, 2022-11-13:
Hello there!
I have run this library successfully inside a docker container on my local machine, where I have had all the administrator privileges. I saved this docker container with other files as a docker image on the docker hub. 
I am trying to work with this docker image in a singularity container on my University's HPC cluster, where, naturally, I do not have any superuser / administrative privileges. So whenever I try to run basic examples such as example_reinforcement_learning or example_turtlebot_navigation, I keep getting the following error -

[image]

Even if I run headlessly, I keep getting the same error. I also tried to run setup.py, and I got the following error - 

[image]

I am pretty sure I am doing many things wrong. However, I cannot figure out where to start and what I am doing wrong. Any help is highly appreciated.

Thank you.

closed 2023-03-18 by stale[bot]

ZXP-S-works, 2023-04-24:
Hi Abetelgeusian,

My collaborator is facing the same issue, did you figure out how to solve the signal 11 issue?

Best,
XP

Abetelgeusian, 2023-05-04:
@ZXP-S-works Unfortunately no. I tried so many different ways but nothing worked.

rsdmse, 2023-10-12:
I am facing the exact same problem in a container. I ran `ldd` on the files in `CoppeliaSim_Edu_V4_1_0_Ubuntu20_04` and some of the dynamic libraries are missing.
```
b0_bubbleRob
	libsodium.so.23 => not found

b0_resolver
	libsodium.so.23 => not found

bubbleRobClient_b0RemoteApi
	libsodium.so.23 => not found

libb0.so
	libsodium.so.23 => not found

libsimExtBlueZero.so
	libsodium.so.23 => not found

libsimExtROS2Interface.so
	libstd_srvs__rosidl_typesupport_cpp.so => not found
	libexample_interfaces__rosidl_typesupport_cpp.so => not found
	libimage_transport.so => not found
	librclcpp.so => not found
	libsensor_msgs__rosidl_typesupport_cpp.so => not found
	librcutils.so => not found
	libtf2_ros.so => not found
	libtf2_msgs__rosidl_typesupport_cpp.so => not found
	librclcpp_action.so => not found
	librcl_action.so => not found
	liblibstatistics_collector.so => not found
	librcl.so => not found
	libstatistics_msgs__rosidl_typesupport_cpp.so => not found
	libtracetools.so => not found
	libgeometry_msgs__rosidl_typesupport_cpp.so => not found
	libstd_msgs__rosidl_typesupport_cpp.so => not found
	libbuiltin_interfaces__rosidl_typesupport_cpp.so => not found

libsimExtROSInterface.so
	libimage_transport.so => not found
	libtf.so => not found
	libroscpp.so => not found
	librosconsole.so => not found
	libroscpp_serialization.so => not found
	librostime.so => not found

libzmq.so
	libsodium.so.23 => not found

libzmq.so.5
	libsodium.so.23 => not found

libzmq.so.5.1.5
	libsodium.so.23 => not found
```
`libsodium.so.23` can be fixed by installing `libsodium23` but I have no idea about the rest. They are not provided by the package manager or `CoppeliaSim_Edu_V4_1_0_Ubuntu20_04`. (I don't know if this is relevant since the OP was able to run it in docker and presumably these libraries should also be missing there.)