stepjam/PyRep #328: libimage_transport.so: cannot open shared object file: No such file or directory

Byte247, 2022-11-25:
Hi, I am trying to create the minimal example: 

from pyrep import PyRep

pr = PyRep()
# Launch the application with a scene file in headless mode
pr.launch('cart_pole.ttt', headless=False) 
pr.start()  # Start the simulation

# Do some stuff

pr.stop()  # Stop the simulation
pr.shutdown()  # Close the application

But I get the following error: 

error: library ( /home/byte/ws/CoppeliaSim/CoppeliaSim_Edu_V4_4_0_rev0_Ubuntu20_04/libsimExtROS.so ) load: "Cannot load library /home/byte/ws/CoppeliaSim/CoppeliaSim_Edu_V4_4_0_rev0_Ubuntu20_04/libsimExtROS.so: (libimage_transport.so: cannot open shared object file: No such file or directory)"
error: library ( /home/byte/ws/CoppeliaSim/CoppeliaSim_Edu_V4_4_0_rev0_Ubuntu20_04/libsimExtROS2.so ) load: "Cannot load library /home/byte/ws/CoppeliaSim/CoppeliaSim_Edu_V4_4_0_rev0_Ubuntu20_04/libsimExtROS2.so: (libstd_srvs__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory)"
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread

Does anyone know how to fix this?

yananliusdu, 2023-02-05:
same errors here....

berniewang8177, 2023-02-07:
This is my error, which is similar.

error: library ( /home/ubuntu/workspace/CoppeliaSim/libsimExtROS.so ) load: "Cannot load library /home/ubuntu/workspace/CoppeliaSim/libsimExtROS.so: (libimage_transport.so: cannot open shared object file: No such file or directory)"
error: library ( /home/ubuntu/workspace/CoppeliaSim/libsimExtROS2.so ) load: "Cannot load library /home/ubuntu/workspace/CoppeliaSim/libsimExtROS2.so: (libstd_srvs__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory)"
[CoppeliaSim:loadinfo]   done.

berniewang8177, 2023-02-07:
> This is my error, which is similar.
>
> error: library ( /home/ubuntu/workspace/CoppeliaSim/libsimExtROS.so ) load: "Cannot load library /home/ubuntu/workspace/CoppeliaSim/libsimExtROS.so: [...]

Hey my friend,

My problem above is solved by using CoppeliaSim of this version, "https://www.coppeliarobotics.com/files/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04.tar.xz"
which is exactly the version linked at the PyRep readme.

My previous CoppeliaSim is from a newer version, which give my that error when I run with headless=True.

closed 2023-05-21 by stale[bot]

18699502246, 2024-10-15:
This is my error, which is similar.
/bin/python3 PyRep/examples/example_panda_end_effector_control.py
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread