stepjam/PyRep #329: qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""

eanswer, 2022-11-28:
I got the error as follow when I tried to run examples:
`
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

[1]    3989205 abort (core dumped)  python example_locobot_stack_cube.py
`

geyuying, 2022-11-29:
Same error

YanjieZe, 2022-12-09:
Same error. Could anyone help possibly

YanjieZe, 2022-12-09:
Hi all, I solve this problem on my remote machine by adding `xvfb-run` before the python command.
For example,
```
xvfb-run python main.py
```
Hope this works for you also.

worasuch, 2023-01-25:
> Hi all, I solve this problem on my remote machine by adding `xvfb-run` before the python command. For example,
> 
> ```
> xvfb-run python main.py
> ```
> 
> Hope this works for you also.

it works for me as well.
how does this command doing actually?

eanswer, 2023-01-25:
It works, thank!

SamNPowers, 2023-04-05:
An alternative solution that worked for me: install opencv-python-headless instead of opencv-python, as suggested: https://github.com/NVlabs/instant-ngp/discussions/300

closed 2023-08-12 by stale[bot]