stepjam/PyRep #335: issues with UR10

yananliusdu, 2023-02-05:
when I build a scene consisting of a UR10 and its gripper for a simple test, but it seems like does not work properly indicating the handle of UR10 can not be found

[image]
[image]

worasuch, 2023-02-12:
I haven't tried UR10 yet. But an error indicates about handle "UR10_target" not "UR10". So, what is "UR10_target" in your sim?

yananliusdu, 2023-02-21:
> I haven't tried UR10 yet. But an error indicates about handle "UR10_target" not "UR10". So, what is "UR10_target" in your sim?

solved, thanks

closed 2023-05-21 by stale[bot]