stepjam/PyRep #345: Does pyrep work with coppeliasim 4.5.1?

nandu-k01, 2023-06-09:


mlzxy, 2023-09-04:
+1 latest Coppeliasim supports Mujoco, which is thought to be better than Bullet by many people.

basab-gupta, 2023-09-08:
I can't be entirely certain, but the numerous segmentation faults lead me to believe that the last version of CoppeliaSim might be compatible with PyRep. I tried to run the examples on CoppeliaSim 4.5.1 but I kept running into segmentation faults. Interestingly enough, I noticed that the code would only break on lines like get and set position but seemed to work fine otherwise. I however, went ahead and installed an older version of CoppeliaSim on Ubuntu 20.04 and that seemed to work with (almost) no issues.

closed 2025-05-06 by stale[bot]