stepjam/PyRep #347: Can't run PyRep on miniconda environment

Harimus, 2023-07-06:
Hello, when I try to run PyREP (the `launch()` ) on my miniconda environment (on Ubuntu 20.04), it seems to crash with the following error:

```
>>> from pyrep import PyRep; pr = PyRep(); pr.launch()
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

In trying to pin-point where the issue is, after checking that the above files (`swrast_dri.so` and `iris_dri.so` ) was indeed available (but in `/usr/lib/x86_64-linux-gnu/dri`) I tried first the solution 2 in this [Stackoverflow answer](https://stackoverflow.com/a/72427700) without solving it. To double check, I tried to run the same process with a python venv instead and it was running without issue, so this seems to be an issue with conda/miniconda, and findingthe correct GL libraries. I'm not sure if I'm asking in the wrong place, but if someone have had similar issue and know how to solve it, It would be greatly help!

Harimus, 2023-07-06:
Nevermind, after running  ` conda install -c conda-forge libffi `  it worked.

closed 2023-07-06 by Harimus

18699502246, 2024-10-08:
I am facing the different issue.
libGL error: MESA-LOADER: failed to open radeonsi: /lib/x86_64-linux-gnu/libLLVM-12.so.1: undefined symbol: ffi_type_sint32, version LIBFFI_BASE_7.0 (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /lib/x86_64-linux-gnu/libLLVM-12.so.1: undefined symbol: ffi_type_sint32, version LIBFFI_BASE_7.0 (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /lib/x86_64-linux-gnu/libLLVM-12.so.1: undefined symbol: ffi_type_sint32, version LIBFFI_BASE_7.0 (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast

PeppaKing8, 2024-12-13:
> I am facing the different issue. libGL error: MESA-LOADER: failed to open radeonsi: /lib/x86_64-linux-gnu/libLLVM-12.so.1: undefined symbol: ffi_type_sint32, version LIBFFI_BASE_7.0 (search paths [...]

Same issue... did you fixed it? @18699502246