stepjam/PyRep #356: [Question]: Steps to support latest CoppeliaSim version

wpumacay, 2023-09-14:
Hi there, thanks for the great library :smiley: . I was wondering what would be the steps required for adding support for the latest version of CoppeliaSim for PyRep. There's better URDF support and some other improvements that might be great to have when using PyRep.

I took a look at how CFFI is used to expose bindings for the C API, and I'd be great if you could clarify or provide some extra information:

- Is the `sim.h` file (used for the CFFI build step) generated from the `simLib.h` file in the CoppeliaSim folder?. I noticed these are in one-to-one correspondance, the difference just being on how they define the API. If that's the case, could I just generate (maybe manually) a similar `sim.h` file from the latest `simLib.h` (from CoppeliaSim v4.5.1), and update the cffi build script accordingly?
- Are there any other steps that I should take into account for generating bindings for the latest version of CoppeliaSim?

Maybe I could give it a try if the steps are not quite difficult.

basab-gupta, 2023-09-15:
I would be happy to help with this issue @stepjam and @wpumacay as well! The latest version of Coppeliasim has a better remote API i.e. the ZeroMPQ and also supports Mujoco which would be great to incorporate into PyRepl.

Coppelia, 2023-09-15:
Just for info, next CoppeliaSim release will have much better support for Python. Via ZMQ remote API, but also embedded Python scripts, and a possibility to run CoppeliaSim with Python handling the main thread (basically what PyRep does not from what I understand). Should be out within 4-6 weeks

stepjam, 2023-09-15:
Happy to support transferring PyRep over to use ZMQ  as the backend once the new CoppleiaSim version comes out. I don't have much time these days to write, but I can certainly advise and point people in the right direction :)

stepjam, 2023-09-15:
Just a note that the biggest hurdle last time I looked at upgrading pyrep ~2 years ago, was the new IK system within CoppeliaSim, which required less trivial changes to make PyRep compatible.

Coppelia, 2023-09-15:
@stepjam, I'd be happy to assist where I can

lgzid, 2023-10-09:
I've recently been using reinforcement learning to train a mobile robotic arm, and PyRep has been a great help to me. I'm looking forward to the new version of PyRep. Thank you for your contributions.

lgzid, 2023-10-19:
@Coppelia Hello！
Do you have any relevant examples to learn ZQM combined with Python API simulation

Coppelia, 2023-10-19:
@lgzid hi, not sure what you are referring to with your question.. ;) Please wait 1-3 days, and we will have CoppeliaSim V4.6 released. It has a much better Python support/connectivity/reactivity with:

- main client application
- embedded scripts
- ZeroMQ remote API clients

lgzid, 2023-10-19:
@Coppelia Thank you for your reply！
Looking forward to the new version very much

Coppelia, 2023-11-10:
@stepjam, wanted to contact you as soon as last few major bugs in V4.6 were fixed, which seems to be the case now. I'd be happy to help out, especially porting existing scenes (e.g. object names for handle retrieval are deprecated in favor of path/alias or IK, collections, etc. should now exclusively be set-up programmatically, not via the GUI)

Jrobo, 2024-01-18:
After running simple example program RL bench, copelliasim is opening with scene, but after 1-2 second it closed with the following error:
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex
any way to solve it?

wpumacay, 2024-01-19:
Uhmmmm, that message appears usually after CoppeliaSim is terminated via pyrep shutdown. Are you using the version of Coppeliasim supported by PyRep? (v4.1), or are you using another version?

Jrobo, 2024-01-21:
I am using version 4.1.

wpumacay, 2024-01-22:
@Jrobo , could you create a separate issue?. In the meantime, are you able to run the `example_youbot_navigation.py` example? That at least should show something

stepjam, 2024-01-22:
Just an FYI that I started the work on the upgrade to 4.6 on the weekend. I expect to have a beta version (which does not include IK) by end of next weekend.
Cheers

vonHartz, 2025-01-16:
@stepjam What's the state of the 4.6 Beta? Is it functional?

ubless607, 2025-01-24:
@stepjam Thank you for developing PyRep.
Will there be any chance to use PyRep on the latest CoppeliaSim v4.9.0?

closed 2025-05-06 by stale[bot]