stepjam/PyRep #357: Feature Request: Improved Documentation and more examples

basab-gupta, 2023-09-18:
Hello @stepjam and the PyRep community,

I've been actively using PyRep and have noticed a couple of areas where I believe I can contribute to improving the project:

1. **Documentation Enhancement:**
   - Issue: libcoppeliaSim.so.1: cannot open shared object file: No such file or directory 
   - Proposed Solution: It seems like some versions of Coppeliasim do not have the symbolic link libcoppeliaSim.so.1 that points to the libcoppeliaSim.so file. I would like to briefly add the steps required to fix this issue by providing intructions needed to generate this file manually. @stepjam Do you think it makes sense to directly add this to the installation readme file?
  

2. **More Examples:**
   - Feature: Adding examples of integrating LiDAR sensors into PyRep simulations.
   - Purpose: Currently, due to PyRep's reliance on the legacy API, users are unable to directly access certain plugins, including simVision, from PyRep. The simVision plugin for instance, has functions that would let us directly access the data from the in-built Veldoyne sensors in Coppeliasim. I would like to add a small working example of a scene where we extract the lidar data from one of the built-in Lidar sensors by using the `script_call` function in PyRep to call the associated Lua child scripts.

3. **scene_panda_reach_target.ttt file**
   - Feature: Change the variable names in the scene file to avoid the errors described in issue #280 

I'd appreciate your feedback on these proposed contributions. Please let me know your thoughts, and if you have any specific guidelines or suggestions.

Thank you for maintaining this amazing repository! I really look forward to contributing to it. :)

stepjam, 2023-09-28:
Hi Basab,

Thansk for the offer to contribute. There is some discussion ongoing (#356) to get PyRep compatible with the most recent version of CoppeliaSim. Once the new version comes out, then helping with the upgrade of PyRep would be great :smiley: . This would then enable the use of the simVision plugin you mentioned too.

stepjam, 2023-09-28:
As I just realise you already commented in #356 :smile:

closed 2025-04-26 by stale[bot]