stepjam/PyRep #364: Can I run PyRep on Ubuntu 22.04

SevenFo, 2023-11-29:
The minimum CoppeliaSim version supported on Ubuntu 22.04 is CoppeliaSim 4.3.0, which isn't compatible with PyRep, as PyRep requires version 4.1.

When I execute the example script python example_baxter_pick_and_pass.py using CoppeliaSim 4.3.0, it appears that CoppeliaSim launches normally. However, I've noticed some logs within CoppeliaSim. 
```
[sandboxScript:info] Simulator launched, welcome!
[WebSocket remote API server@addOnScript:error] error loading module 'lpeg' from file '/opt/CoppeliaSim_Edu_V4_3_0/luar/lpeg.so':
    /opt/CoppeliaSim_Edu_V4_3_0/luar/lpeg.so: undefined symbol: lua_gettop
stack traceback:
    [C]: in ?
    [C]: in field 'require'
    /opt/CoppeliaSim_Edu_V4_3_0/lua/sim.lua:27: in function 'require'
    /opt/CoppeliaSim_Edu_V4_3_0/luar/org/conman/cbor.lua:99: in main chunk
    [C]: in field 'require'
    /opt/CoppeliaSim_Edu_V4_3_0/lua/sim.lua:27: in function 'require'
    [string "WebSocket remote API server@addOnScript"]:113: in function 'sysCall_init'
[ZMQ remote API server@addOnScript:error] error loading module 'lpeg' from file '/opt/CoppeliaSim_Edu_V4_3_0/luar/lpeg.so':
    /opt/CoppeliaSim_Edu_V4_3_0/luar/lpeg.so: undefined symbol: lua_gettop
stack traceback:
    [C]: in ?
    [C]: in field 'require'
    /opt/CoppeliaSim_Edu_V4_3_0/lua/sim.lua:27: in function 'require'
    /opt/CoppeliaSim_Edu_V4_3_0/luar/org/conman/cbor.lua:99: in main chunk
    [C]: in field 'require'
    /opt/CoppeliaSim_Edu_V4_3_0/lua/sim.lua:27: in function 'require'
    [string "ZMQ remote API server@addOnScript"]:124: in function 'sysCall_init'
[ZMQ remote API server@addOnScript:error] 39: in simZMQ.close@simExtZMQ: read in arg 1 (socket): expected string
stack traceback:
    [C]: in field '?'
    /opt/CoppeliaSim_Edu_V4_3_0/lua/simZMQ.lua:39: in function 'simZMQ-typecheck.close'
    [string "ZMQ remote API server@addOnScript"]:141: in function 'sysCall_cleanup'
[sandboxScript:info] Simulation started.
```
I'm uncertain whether this behavior is normal or if it indicates an error.

lakshitadodeja, 2023-12-13:
Hi ! How did you install the 4.3 version of CoppeliaSim ? I am only able to see 4.5 here - https://www.coppeliarobotics.com/previousVersions

SevenFo, 2023-12-14:
> Hi ! How did you install the 4.3 version of CoppeliaSim ? I am only able to see 4.5 here - https://www.coppeliarobotics.com/previousVersions

It's strange. There was an installation package for this version a few days ago.  
Maybe have been removed.

I later installed version 4.1 into a docker image of Ubuntu 20.04. It can work normally.

SevenFo, 2023-12-14:
Close this issue as there's no 4.3 version of CoppeliaSim

closed 2023-12-14 by SevenFo