stepjam/PyRep #374: errors happen when running the example

nuclear-missile, 2024-04-28:
I cd the example folder and run examples, they throw different errors

python3 few_shot_rl.py or multi_task_rl.py or rlbench_gym.py or rearrangement_challenge.py
Reset Episode
['reach the orange target', 'touch the orange ball with the panda gripper', 'reach the orange sphere']
Traceback (most recent call last):
  File "few_shot_rl.py", line 53, in <module>
    obs, reward, terminate = task.step(action)
  File "/home/yczhang/hzj_diff/RLBench/rlbench/task_environment.py", line 99, in step
    self._action_mode.action(self._scene, action)
  File "/home/yczhang/hzj_diff/RLBench/rlbench/action_modes/action_mode.py", line 35, in action
    self.arm_action_mode.action(scene, arm_action, ignore_collisions)
TypeError: action() takes 3 positional arguments but 4 were given
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex

Error: signal 11:

python3 imitation_learning.py 
imitation_learning.py:39: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  demos = np.array(demos).flatten()
'training' iteration 0
'training' iteration 1
'training' iteration 2
'training' iteration 3
'training' iteration 4
'training' iteration 5
'training' iteration 6
'training' iteration 7
'training' iteration 8
'training' iteration 9
'training' iteration 10
'training' iteration 11
'training' iteration 12
'training' iteration 13
'training' iteration 14
'training' iteration 15
'training' iteration 16
'training' iteration 17
'training' iteration 18
'training' iteration 19
'training' iteration 20
'training' iteration 21
'training' iteration 22
'training' iteration 23
'training' iteration 24
'training' iteration 25
'training' iteration 26
'training' iteration 27
'training' iteration 28
'training' iteration 29
'training' iteration 30
'training' iteration 31
'training' iteration 32
'training' iteration 33
'training' iteration 34
'training' iteration 35
'training' iteration 36
'training' iteration 37
'training' iteration 38
'training' iteration 39
'training' iteration 40
'training' iteration 41
'training' iteration 42
'training' iteration 43
'training' iteration 44
'training' iteration 45
'training' iteration 46
'training' iteration 47
'training' iteration 48
'training' iteration 49
'training' iteration 50
'training' iteration 51
'training' iteration 52
'training' iteration 53
'training' iteration 54
'training' iteration 55
'training' iteration 56
'training' iteration 57
'training' iteration 58
'training' iteration 59
'training' iteration 60
'training' iteration 61
'training' iteration 62
'training' iteration 63
'training' iteration 64
'training' iteration 65
'training' iteration 66
'training' iteration 67
'training' iteration 68
'training' iteration 69
'training' iteration 70
'training' iteration 71
'training' iteration 72
'training' iteration 73
'training' iteration 74
'training' iteration 75
'training' iteration 76
'training' iteration 77
'training' iteration 78
'training' iteration 79
'training' iteration 80
'training' iteration 81
'training' iteration 82
'training' iteration 83
'training' iteration 84
'training' iteration 85
'training' iteration 86
'training' iteration 87
'training' iteration 88
'training' iteration 89
'training' iteration 90
'training' iteration 91
'training' iteration 92
'training' iteration 93
'training' iteration 94
'training' iteration 95
'training' iteration 96
'training' iteration 97
'training' iteration 98
'training' iteration 99
Done
[CoppeliaSim:loadinfo]   done.
QObject::~QObject: Timers cannot be stopped from another thread

I am sure the copilia and the pyrep are truely installed. I try to run the example in pyrep, it cancorrectly run, but "QObject::~QObject: Timers cannot be stopped from another thread" at the end. I fund it means the python thread finish but copilia not close. in the RLbench examples it can't even begin. 

Here is my enviroment:
Ubuntu 20.04.6 LTS
CUDA Version: 12.4
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
CoppeliaSim_Edu_V4_1_0_Ubuntu20_04
PyRep                         4.1.0.3
rlbench                       1.2.0 
Python 3.8.19

nuclear-missile, 2024-04-28:
my zshrc contains
export QT_PLUGIN_PATH="/usr/lib/x86_64-linux-gnu/qt5/plugins"
export COPPELIASIM_ROOT=/home/yczhang/hzj_diff/PyRep/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT
export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT
export DPYTHON_EXECUTABLE=/usr/bin/python3

Benxiaogu, 2024-10-20:
Hello!
I'm having the same problem as you and I can't solve the problemeven though I've tried multiple things.
I would like to get the way to solve the problem if you have it work!
Thank you!

> I cd the example folder and run examples, they throw different errors
> 
> python3 few_shot_rl.py or multi_task_rl.py or rlbench_gym.py or rearrangement_challenge.py Reset Episode ['reach the orange target', 'touch the orange ball with the panda gripper', 'reach the orange sphere'] Traceback (most recent call last): File "few_shot_rl.py", line 53, in obs, reward, terminate = task.step(action) File "/home/yczhang/hzj_diff/RLBench/rlbench/task_environment.py", line 99, in step self._action_mode.action(self._scene, action) File "/home/yczhang/hzj_diff/RLBench/rlbench/action_modes/action_mode.py", line 35, in action self.arm_action_mode.action(scene, arm_action, ignore_collisions) TypeError: action() takes 3 positional arguments but 4 were given QObject::~QObject: Timers cannot be stopped from another thread QMutex: destroying locked mutex
> 
> Error: signal 11:
> 
> python3 imitation_learning.py imitation_learning.py:39: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray. demos = np.array(demos).flatten() 'training' iteration 0 'training' iteration 1 'training' iteration 2 'training' iteration 3 'training' iteration 4 'training' iteration 5 'training' iteration 6 'training' iteration 7 'training' iteration 8 'training' iteration 9 'training' iteration 10 'training' iteration 11 'training' iteration 12 'training' iteration 13 'training' iteration 14 'training' iteration 15 'training' iteration 16 'training' iteration 17 'training' iteration 18 'training' iteration 19 'training' iteration 20 'training' iteration 21 'training' iteration 22 'training' iteration 23 'training' iteration 24 'training' iteration 25 'training' iteration 26 'training' iteration 27 'training' iteration 28 'training' iteration 29 'training' iteration 30 'training' iteration 31 'training' iteration 32 'training' iteration 33 'training' iteration 34 'training' iteration 35 'training' iteration 36 'training' iteration 37 'training' iteration 38 'training' iteration 39 'training' iteration 40 'training' iteration 41 'training' iteration 42 'training' iteration 43 'training' iteration 44 'training' iteration 45 'training' iteration 46 'training' iteration 47 'training' iteration 48 'training' iteration 49 'training' iteration 50 'training' iteration 51 'training' iteration 52 'training' iteration 53 'training' iteration 54 'training' iteration 55 'training' iteration 56 'training' iteration 57 'training' iteration 58 'training' iteration 59 'training' iteration 60 'training' iteration 61 'training' iteration 62 'training' iteration 63 'training' iteration 64 'training' iteration 65 'training' iteration 66 'training' iteration 67 'training' iteration 68 'training' iteration 69 'training' iteration 70 'training' iteration 71 'training' iteration 72 'training' iteration 73 'training' iteration 74 'training' iteration 75 'training' iteration 76 'training' iteration 77 'training' iteration 78 'training' iteration 79 'training' iteration 80 'training' iteration 81 'training' iteration 82 'training' iteration 83 'training' iteration 84 'training' iteration 85 'training' iteration 86 'training' iteration 87 'training' iteration 88 'training' iteration 89 'training' iteration 90 'training' iteration 91 'training' iteration 92 'training' iteration 93 'training' iteration 94 'training' iteration 95 'training' iteration 96 'training' iteration 97 'training' iteration 98 'training' iteration 99 Done [CoppeliaSim:loadinfo] done. QObject::~QObject: Timers cannot be stopped from another thread
> 
> I am sure the copilia and the pyrep are truely installed. I try to run the example in pyrep, it cancorrectly run, but "QObject::~QObject: Timers cannot be stopped from another thread" at the end. I fund it means the python thread finish but copilia not close. in the RLbench examples it can't even begin.
> 
> Here is my enviroment: Ubuntu 20.04.6 LTS CUDA Version: 12.4 QMake version 3.1 Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu CoppeliaSim_Edu_V4_1_0_Ubuntu20_04 PyRep 4.1.0.3 rlbench 1.2.0 Python 3.8.19

closed 2025-05-06 by stale[bot]