stepjam/PyRep #383: PR for RuntimeWarning: divide by zero encountered in scalar divide

auee028, 2024-08-01:
I got an error of zero division error as below:
```
/home/user/.../anaconda3/envs/.../lib/python3.9/site-packages/pyrep/robots/configuration_paths/arm_configuration_path.py:172: RuntimeWarning:

divide by zero encountered in scalar divide

/home/user/.../anaconda3/envs/.../lib/python3.9/site-packages/pyrep/robots/configuration_paths/arm_configuration_path.py:181: RuntimeWarning:

invalid value encountered in multiply
```

To avoid such error, it seems necessary to do something like to change line 172 from `t = (pos - lengths[i]) / (lengths[i + 1] - lengths[i])` to `t = (pos - lengths[i]) / (lengths[i + 1] - lengths[i]) if (lengths[i + 1] - lengths[i]) else 0`.

MurakamiNatsuki, 2024-08-22:
Hi! I got same error.
Did you manage to solve it as in your example? Or is it still unresolved?

MurakamiNatsuki, 2024-08-23:
Thank you for your prompt reply! 
I suspected it might be an issue with the version, but it seems that’s not the case.
 If you find out anything else, please let me know. I’ll also do some research on my end.

auee028, 2024-08-23:
Thanks, I will also try to figure out and update if I find a solution.

closed 2025-05-06 by stale[bot]